thebradleysanders/ESPHome-Arduino-Port-Expander #1: Not working with Mega2560

DamianBroj, 2021-09-20:
Hi Bradley, i have just tried to upload .ino and .h into my Mega2560 + Nodemcu and it is not flying together. Is the current commit the working one?

thebradleysanders, 2021-09-20:
I see, yes the current commit is what I have working. Try changing your I2C pins:

```
i2c:
  - id: i2c_component1
    sda: 2
    scl: 4
```

DamianBroj, 2021-09-20:
So the I2C seems not be an issue. Issue is that port numbering is not matching. Whhat is the pin reference that supposed to be used in esphome vs the labeling on mega board?

thebradleysanders, 2021-09-20:
I see, port 22 on the Arduino Mega should be referenced as 22 in ESPHome. it's 1:1 pin mapping. I am not sure what can be causing this. I will look into it.

thebradleysanders, 2021-09-20:
Do pins 0-13 map correctly?

DamianBroj, 2021-09-20:
unfortunately i can't trigger a change in any of the ports. can we try to solve it off the git comments and post conclusions here afterwards? you can reach me on linkedin, facebook or slack

DamianBroj, 2021-09-20:
I am using A3-A13 as binary outputs and D29-D36 + A2 as binary sensors. Using files from ESPhome site (that support Pro mini board) i am well able to trigger all of the outputs using node and use A2 switch. D29-D36 not reacting.

After i upload your .h file i am still able to toggle outputs, but A2 input is not working.

After i upload .ino Mega is not toggling outputs (ESPhome shows everything properly, but no effect on arduino's side)

DamianBroj, 2021-09-22:
Ok, i have an update as i figured out what is causing issue, but not so well prepared to fix that.

So the part that is making communication between ESP and Mega broken (Arduino not getting commands in loop) is the following:

#define CMD_ANALOG_READ_A15 10111

This part of code exists as in .h and in .ino files. 

Basically any binary value other than the original one, so 15 (original code defined A0 as binary 8 and A7 as binary 15 registering 8 analogs) is breaking the communication. Keeping it not changed vs original code and applying all other changes that you did, so extending buffer and enabling reading it properly, are working well.

@thebradleysanders i am looking forward to have this thing solved, i can support more if needed as i have a test hardware up and running

DamianBroj, 2021-09-22:
Could you please comment here with the information what is that part of the code doing and what is the explanation of assigning to those defines certain values? The original code (that is not breaking communication with mega as well) below:

`#define CMD_ANALOG_READ_A0 0b1000 // 0x8` - what the binary 8 stands for? is it the PCINT pin number in bin?
`// ....`
`#define CMD_ANALOG_READ_A7 0b1111 // 0xF` - the same question as above?
`//`
`#define CMD_SETUP_ANALOG_INTERNAL 0x10`
`#define CMD_SETUP_ANALOG_DEFAULT 0x11` - what the ...INTERNAL and ...DEFAULT defines are used for and what is the reason behind hex 10 & 11 numbers there?

as for now i am referencing 2 sources to get alignemnt on identifying the meaning of those values, after you help me with that i can move with further things.

Pro mini pinout where i could find "8" assigned to A0 pin that is PCINT number:
[https://4.bp.blogspot.com/-k4j06ktCXbE/WaptVL63NuI/AAAAAAAACPY/_AwacPBjBaMfrmOTsKPetZi5_BgKATB_wCLcBGAs/s1600/ProMiniPinout.png](url)

Documentation of analogread() function saying that for mega2560 analog pins that are usable are A0-A14, not A15:
[https://www.arduino.cc/reference/en/language/functions/analog-io/analogread/](url)

thebradleysanders, 2021-09-22:
#define CMD_ANALOG_READ_A0 0b1000 // 0x8 - what the binary 8 stands for? is it the PCINT pin number in bin? **Yes**
#define CMD_ANALOG_READ_A7 0b1111 // 0xF

I am not sure how these are used. My thought when changing them, I thought they were used to define what pins fell into the read_analog command list CMD_ANALOG_READ_A0 - CMD_ANALOG_READ_A7

I am still looking into this.

thebradleysanders, 2021-09-22:
Documentation of analogread() function saying that for mega2560 analog pins that are usable are A0-A14, not A15:
https://www.arduino.cc/reference/en/language/functions/analog-io/analogread/

I was not aware of this, it will be adjusted. Thank you.

DamianBroj, 2021-09-23:
Hold on with changing those pcint values as on mega board there is a different setup of such pins. 
On the original pro mini there were analog A0-A5 pins having a PCINT reference set (8-13), A6-A7 not. In code there is A7 referenced as PCINT15 that basically doesn't exist for that board (confused) but analogread function doesn't break the communication because it covers A0-A5 and RST (PCINT14). Refer to the below:
https://aws1.discourse-cdn.com/arduino/original/4X/f/7/e/f7eda2a9afb7e1ea2c468630cafb4cf49954e127.jpeg

Looking at mega2560 there is a different setup of PCINT pins that covers only half of the analog pin set. From the whole set of analog pins A0-A15 only A8-A15 are the interrupt capable with the range of PCINT16-23. Referring to that:
https://duino4projects.com/wp-content/uploads/2013/04/ardunio_mega_pinout.jpg?_gl=1*n0c386*_ga*YW1wLVNHWWlJMzQzcndzSXB6MWdyOFRxQ1U4WXd5SU1nMEdab2RTQkhNQXdBenRUSFdlbGhTUjlERklsRnpSTXpNX3U.

So it can be more tricky thing to get it done 100% as invented for pro mini. I am not an expert here but looking at the fact that those analog pins are referred as PCINT ones then my believe is that the code is doing something with the interruption functionality that in case of manipulating there triggers interrupt on launch making Arduino loop not even starting.

marceloabranco, 2022-01-11:
Hi Bradley. Its a great job this one you did. I'm able to use arduino mega to control lamps and switches all over my home.
I'm just wondering, for an another project, would it be possible to control servos through the mega or uno using this esphome port expander?
I'm still a newbie.
Thankyou

thebradleysanders, 2022-01-12:
Thank you for your positive feedback. Good idea. You should be able to with the digital output pins.what servo are you using? I may be able to help.

closed 2023-03-17 by thebradleysanders