thebradleysanders/ESPHome-Arduino-Port-Expander #4: Help converting atmega32u4

NonaSuomy, 2022-10-28:
Hi @thebradleysanders 
I see you converted the code for the mega any idea how to convert it for the 32u4? 

https://github.com/thebradleysanders/ESPHome-Arduino-Port-Expander/blob/main/arduino_port_expander.h#L34

How do you get the right binary values for the pins?

```
01 (INT.6/AIN0) PE6
02 UVcc
03 D
04 D+
05 UGnd
06 UCap
07 VBus
08 (SS/PCINT0) PB0
09 (PCINT1/SCLK) PB1
10 (PDI/PCINT2/MOSI) PB2
11 (PDO/PCINT3/MISO) PB3
12 (PCINT7/OC0A/OC1C/RTS) PB7
13 RESET
14 VCC
15 GND
16 XTAL2
17 XTAL1
*18 (OC0B/SCL/INT0) PD0
*19 (SDA/INT1) PD1
20 (RXD1/INT2) PD2
21 (TXD1/INT3) PD3
22 (XCK1/CTS) PD5
23 GND
24 AVCC
25 PD4 (ICP1/ADC8)
26 PD6 (T1/OC4D/ADC9)
27 PD7 (T0/OC4D/ADC10)
28 PB4 (PCINT4/ADC11)
29 PB5 (PCINT5/OC1A/OC4B/ADC12)
30 PB6 (PCINT6/OC1B/OC4B/ADC13)
31 PC6 (OC3A/OC4A)
32 PC7 (ICP3/CLK0/OC4A)
33 PE2 (HWB)
34 VCC
35 GND

Analog
32u4                         324p                                          Not defined
36 PF7 (ADC7/TDI)   22 ADC7 
37 PF6 (ADC6/TDO) 19 ADC6 
38 PF5 (ADC5/TMS) 28 PC5 (ADC5/SCL/PCINT13)    26 PC3 (ADC3/PCINT11)
39 PF4 (ADC4/TCK)  27 PC4 (ADC4/SDA/PCINT12)   25 PC2 (ADC2/PCINT10)
40 PF1 (ADC1)          22 PC1 (ADC1/PCINT9)
41 PF0 (ADC0)          23 PC0 (ADC0/PCINT8)

42 AREF
43 GND
44 AVCC
```

Thank you for any help.

thebradleysanders, 2022-10-28:
I will work on this for you, not sure how long it will take, could be quick. It’s been a while since I did this so I’ll have to refresh myself, thanks for the project! What exact board do you have?Sent from my iPhone

https://github.com/thebradleysanders/ESPHome-Arduino-Port-Expander/blob/main/arduino_port_expander.h#L34

thebradleysanders, 2022-10-28:
Better yet, it should work, the only thing that shouldn't is the analog pins, can you try and and let me know if the digital pins work? I don't have a board to test.

NonaSuomy, 2022-10-28:
[image]
[image]

NonaSuomy, 2022-10-28:
Is it possible to convert the analog pin defined part to use Arduino A0 etc internal pin definitions so these binary values wouldn't be required?

NonaSuomy, 2022-10-29:
With the default code base:
```
GND                                                            VCC +5V
00 B0  (00)                                                   F0 21 A00 (A1 Value:538     Idle: 96)
01 B1  (01)                                                   F1 20 A01 (A0 Value:32864 Idle: 96)
02 B2  (02)                                                   F4 19 A02 (?? Output 0)
03 B3  (03)                                                   F5 18 A03 (?? Output 0)
04 B7  (04)                                                   F6 17 A04 (?? Output 0)
24 E8  (?? No response)     (Interior)  AREF AF 
05 D0 (SCL)                                                  F7 16 A05 (?? Output 0)
06 D1 (SDA)                                                 B6 15 A06 (?? Output 0)
07 D2 (07)                                                    B5 14 A07 (?? Output 0)
08 D3 (08)                                                    B4 13 A08 (?? Output 0)
09 C6 (09)                                                    D7 12 A09 (?? Output 0)
10 C7 (10)                                                    D6 11 A10 (?? Output 0)
23 D3 (?? No response)                               D4 22 A11 (?? Output 0)
+5V 
GND 
RST  
 ```

Stuff in brackets confirmed pins applied 2.64V with the 10K Pot to the analog pins. 7.09KOhm~

NonaSuomy, 2022-10-30:
I'm not really sure what I'm doing but I got more than 0's by playing with the binary values. In the new Arduino IDE 2.0 I could hover over the pin values and then it would show me the correct pin values:

`
//A11 22 0x16 10110;
//A0 21 0x15 10101;
//A1 20 0x14 10100;
//A2 19 0x13 10011;
//A3 18 0x12 10010;
//A4 17 0x11 10001;
//A5 16 0x10 10000;
//A6 15 0xf 1111;
//A7 14 0xe 1110;
//A8 13 0xd 1101;
//A9 12 0xc 1100;
//A10 10 0xa 1010;
`
The values are dumping all over the place now so I'm sure I'm messing up the way it processes the analog values now with its bitwise operation.

Only pin F5 connected:
```
[01:52:11][D][sensor:127]: 'Analog A0': Sending state 32.00000  with 0 decimals of accuracy
[01:52:12][D][sensor:127]: 'Analog A1': Sending state 32.00000  with 0 decimals of accuracy
[01:52:12][D][sensor:127]: 'Analog A2': Sending state 32.00000  with 0 decimals of accuracy
[01:52:12][D][sensor:127]: 'Analog A3': Sending state 32.00000  with 0 decimals of accuracy
[01:52:13][D][sensor:127]: 'Analog A4': Sending state 1023.00000  with 0 decimals of accuracy
[01:52:13][D][sensor:127]: 'Analog A5': Sending state 977.00000  with 0 decimals of accuracy
[01:52:13][D][sensor:127]: 'Analog A6': Sending state 1023.00000  with 0 decimals of accuracy
[01:52:13][D][sensor:127]: 'Analog A7': Sending state 1023.00000  with 0 decimals of accuracy
[01:52:13][D][sensor:127]: 'Analog A8': Sending state 943.00000  with 0 decimals of accuracy
[01:52:13][D][sensor:127]: 'Analog A9': Sending state 896.00000  with 0 decimals of accuracy
[01:52:13][D][sensor:127]: 'Analog A10': Sending state 894.00000  with 0 decimals of accuracy
[01:52:13][D][sensor:127]: 'Analog A11': Sending state 848.00000  with 0 decimals of accuracy
```
No pins connected:
```
[01:54:02][D][sensor:127]: 'Analog A0': Sending state 32.00000  with 0 decimals of accuracy
[01:54:03][D][sensor:127]: 'Analog A1': Sending state 32.00000  with 0 decimals of accuracy
[01:54:03][D][sensor:127]: 'Analog A2': Sending state 32.00000  with 0 decimals of accuracy
[01:54:03][D][sensor:127]: 'Analog A3': Sending state 32.00000  with 0 decimals of accuracy
[01:54:03][D][sensor:127]: 'Analog A4': Sending state 32.00000  with 0 decimals of accuracy
[01:54:04][D][sensor:127]: 'Analog A5': Sending state 143.00000  with 0 decimals of accuracy
[01:54:04][D][sensor:127]: 'Analog A6': Sending state 159.00000  with 0 decimals of accuracy
[01:54:04][D][sensor:127]: 'Analog A7': Sending state 150.00000  with 0 decimals of accuracy
[01:54:04][D][sensor:127]: 'Analog A8': Sending state 146.00000  with 0 decimals of accuracy
[01:54:04][D][sensor:127]: 'Analog A9': Sending state 32.00000  with 0 decimals of accuracy
[01:54:04][D][sensor:127]: 'Analog A10': Sending state 151.00000  with 0 decimals of accuracy
[01:54:04][D][sensor:127]: 'Analog A11': Sending state 146.00000  with 0 decimals of accuracy
```
Full code
`
/*
Ports:
	0 0 .. 13 13
	A0: 14, A1: 15, A2: 16, A3: 17: A4: 18: A5: 19: A6: 20, A7: 21
	port bits: 5 ... 0..32
	0:   0: 00000
	1:	 1:	00001
	A7: 21: 10101
*/

#include <Arduino.h>
#include <Wire.h>

//#define DEBUG 1 // remove debug so pin 0 and 1 can be used for IO
//#define DEBUG_READ 1
#define I2C_ADDRESS 9

void onRequest();
void onReceive(int);

[... 198 lines cut ...]
    break;
  case CMD_SETUP_PIN_INPUT_PULLUP:
    pinMode(pin, INPUT_PULLUP);
#ifdef DEBUG
    Serial.print(F("Pin "));
    Serial.print(pin);
    Serial.println(F("INPUT PULLUP"));
#endif
    break;
  case CMD_SETUP_ANALOG_INTERNAL:
    analogReference(INTERNAL);
//#ifdef DEBUG
    //Serial.println(F("Analog reference INTERNAL"));
//#endif
    break;
  case CMD_SETUP_ANALOG_DEFAULT:
    analogReference(DEFAULT);
//#ifdef DEBUG
    //Serial.println(F("Analog reference DEFAULT"));
//#endif
    break;
  }
}

`