thebradleysanders/ESPHome-Arduino-Port-Expander #6: Relays not activating

Keshin1988, 2023-05-05:
hello im not sure if you can help but im only amateur at best with coding. attached i will show the coding i used 

Problem (if i run the blink sketch with the relays on the arduino they work) did this to confirm boards are functioning correctly
when i connect everything up and run it through home assistant (esphome) it shows i2c has been found, and arduino serial logger show pin 2 pulled high, pin 2 pulled low when switch is toggled, however the relay does not do anything.
if i connect a voltage meter whilst all is still plugged in i get 4.83 volts and when the switch is toggled to on i get 0.10 volts

attached is both your code and the esphome code (example i am using to test before changing it)

### **Your code**
```
/*
Ports:
  0 0 .. 13 13 .. 53 53
  Uno: A0: 14, A1: 15, A2: 16, A3: 17: A4: 18: A5: 19: A6: 20, A7: 21
  Mega: A0: 54, A1: 55, A2: 56, A3: 57: A4: 58: A5: 59: A6: 60, A7: 61...
  port bits: 5 ... 0..32
  0:   0: 00000
  1:   1: 00001
  A7: 21: 10101
*/

#include <Arduino.h>
#include <Wire.h>

#define DEBUG // remove debug so pin 0 and 1 can be used for IO
//#define DEBUG_READ // more advanced debuging

#define I2C_ADDRESS 8 //i2c starts at pin 8

void onRequest();
void onReceive(int);

void setup()
{
#ifdef DEBUG
  Serial.begin(115200);
  Serial.println(F("Init "));
#endif

#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
    analogReference(INTERNAL1V1);
#endif
#if defined(__AVR_ATmega328P__) || defined(__AVR_ATmega168__)
    analogReference(INTERNAL);
#endif

  Wire.begin(I2C_ADDRESS);
  Wire.onRequest(onRequest);
  Wire.onReceive(onReceive);

#ifdef DEBUG
  Serial.println(F("Wire ok"));
#endif
}

void loop()
{
  //int temp = analogRead(A1);
  //Serial.println(temp);
}

volatile byte buffer[9];
volatile byte len = 1;

#define DIGITAL_READ(b, pin, mask) \
  if (digitalRead(pin))            \
    buffer[b] |= mask;

void readDigital()
{
[... 380 lines cut ...]
        name: Garage garage
        device_class: garage_door

# define analog sensors
sensor:
  - platform: custom
    lambda: |-
      return {ape_analog_input(expander1, 1),  // 1 = A1
              ape_analog_input(expander1, 2)};
    sensors:
      - name: LM35 Living room temperature
        id: lm35_temp
        filters:
          # update every 60s
          - throttle: 60s
          # LM35 outputs 0.01v per ºC, and 1023 means 3.3 volts
          - lambda: return x * 330.0 / 1023.0;
      - name: Analog A2
        id: analog_a2
        filters:
          - throttle: 2s
```

### **Crude sketch of my circuit**
[image]

Keshin1988, 2023-05-05:
It appears it was a pin 2 issue whether it's with my board or in the coding I'm unsure. Will leave this open in case issues return. Will update after I implement and run it for a few days

oxedgar, 2024-09-08:
I found issue! When ape_analog_input(expander1, 2)}; is listed on esphome, the relay on D2 in not working. Not tested but should be similar on other analog pins. 

Also, i try to use ape_analog_input(expander1, 55)}; to read A2 pin but i never receive values. always 41223.00000  in all pins where is readed supperior as 54.

I'm searching the way to make it working. peraps the develloper can help me to do that?

[Keshin1988](https://github.com/Keshin1988): If  you do not use A2 pin, remove it from you esphome and the relay is work fine!