thebradleysanders/ESPHome-Arduino-Port-Expander #8: custom component no more supported

vitucciog, 2025-03-23:
Hi,
With the recent removal of custom components from Esphome, it is no more possible to compile and install an esphome device including this arduino port expander.
Are you going to convert and release it as external component?

Thanks. Bye,
Vito

webneers, 2025-03-26:
Hi, same problem here. Would really love to have it converted! Thank you so much.

blackscreener, 2025-09-05:
@webneers @vitucciog 
Check out my fork. The code was almost entirely generated by AI; the implemented components are switch and binary sensor. The switch component supports the inverted and interlock functions for controlling roller shutters. After an ESP reset, the component by default turns off all configured outputs.

```
external_components:
  - source:
      type: local
      path: custom_components  
    components: [ arduino_port_expander ]
    refresh: 0s

switch:
  - platform: arduino_port_expander
    arduino_port_expander: ape
    pin: 38
    id: relay_up
    inverted: false
    interlock: [relay_down]
    interlock_wait_time: 500ms
    name: Relay up
    internal: false
    restore_mode: ALWAYS_OFF

  - platform: arduino_port_expander
    arduino_port_expander: ape
    pin: 39
    id: relay_down
    inverted: false
    interlock: [relay_up]
    interlock_wait_time: 500ms
    name: Relay down
    internal: false
    restore_mode: ALWAYS_OFF

binary_sensor:
  - platform: template
    name: "I2C Communication Status"
    id: i2c_communication_status
    device_class: connectivity
    lambda: |-
      return !id(ape).is_failed();
    on_release: 
      then:
        - logger.log: "I2C communication lost, attempting recovery..."
        - delay: 2min
        - if:
            condition:
              binary_sensor.is_off: i2c_communication_status
            then:
              - logger.log: "I2C still down, restarting ESP..."
              - delay: 5s
              - button.press: restart_esp

  - platform: arduino_port_expander
    arduino_port_expander: ape
    pin: 22
    name: "Button up"
    id: button_up
    inverted: true
    internal: false
```